polarfire-soc/icicle-kit-reference-design #1: libero project creation

rroulleau, 2020-09-23:
When running the script to create the project it is looking for F: drive location. 
See the script execution report.

Error: Failure when executing Tcl script. [ Line 48: Internal Error:
Assertion Failed in "F:/release/sn/capture_sjts68_joy_jen_sn/afi/prg/pfsoc_mss/src/pa5socmssApp.cpp", line 192.

[microsemi_ScriptExecutionReport.txt](https://github.com/polarfire-soc/icicle-kit-reference-design/files/5267431/microsemi_ScriptExecutionReport.txt)

hughbreslin, 2020-09-23:
Hey @rroulleau does the path to the script have any spaces in it? I can reproduce a similar failure if I run using a path with a space.

rroulleau, 2020-09-23:
Correct, there was space in the path of the project! Thanks a lot!

closed 2020-09-23 by rroulleau