polarfire-soc/icicle-kit-reference-design #2: update eFP6 to rev B failed

jinchenglee, 2020-12-22:
I've got such a version that I need to update eFP6 to rev. B as suggested in README.md. 

_Follow the steps in this [document](https://download-soc.microsemi.com/FPGA/v12.5/sp1/README.pdf) to program the eFP6 on the Icicle Kit_

I've installed FPExpress v12.5 and downloaded the eFP6_revB_TAG_v4_ppd.job, however, the FPExpress requires the project directory location that I don't have one nor can I find one anywhere. The program shows as below when I try to click "OK". 

[image]

nitindeshpande, 2020-12-22:
"Project location" can be any folder on your machine which FPExpress can use. it is OK to choose the folder being chosen as shown in the image. 
 Please also note that, you must use 12.5 SP1.

jinchenglee, 2020-12-22:
It turns out that Windows 10 Pro seems have made the Downloads directory special (I don't probe further). Using that directory will report error. Change to a normal directory, say C:\blahblah works. Thanks.

closed 2020-12-22 by jinchenglee