polarfire-soc/icicle-kit-reference-design #3: Cannot update "SoC-FPGA + zero-stage bootloader FlashPro Express programming file" with modern FlashPro Express software

aleixrocks, 2020-12-22:
When trying to update the "SoC-FPGA + zero-stage bootloader FlashPro Express programming file" (https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/boards/mpfs-icicle-kit-es/updating-icicle-kit/updating-icicle-kit-design-and-linux.md) for eMMC on my MPFS-ICICLE-KIT-ES-FP5 using FlashPro Express 12.6 I get the following error:

```
Software Version: 12.900.20.24
PPD file '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/update_2020_11/MPFS_ICICLE/MPFS_ICICLE.ppd' has been loaded successfully.
DESIGN : MPFS_ICICLE_eMMC;  CHECKSUM : F839;  PDB_VERSION : 1.0
￼Warning: Original Embedded FP6 '18B74665' is found.  It must be updated to work with this version of the software.  Please refer to software release notes on how to update it.
￼Error: Failed to open the project file '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/update_2020_11/MPFS_ICICLE/MPFS_ICICLE.pro'.
```

Following the instructions in this document's `Embedded FlashPro 6` section, I try to upload the 

  https://download-soc.microsemi.com/FPGA/v12.5/sp1/eFP6_revB_TAG_v4_ppd.job

file using the external FlashPro 5 hardware and FlashPro Express 12.6 software. However, I got the error:

```
Software Version: 12.900.20.24
PPD file '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.ppd' has been loaded successfully.
DESIGN : FlashPro6;  CHECKSUM : A8EA;  PDB_VERSION : 1.0
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
programmer 'S20053Q3C8' : FlashPro5
Opened '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.pro'
￼Warning: FlashPro5 TCK was set to 4000000, but has been reduced to 30 to meet the minimum max TCK of the chain.
programmer 'S20053Q3C8' : Scan Chain...
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
programmer 'S20053Q3C8' : Check Chain...
￼Error: programmer 'S20053Q3C8' : Device 1: Found: MPFS250(T_ES|TS_ES), Expected: M2S025
￼Error: programmer 'S20053Q3C8' : Scan and Check Chain FAILED.
￼Error: Failed to run Action.
```

When using FlashPro Express 12.5 software I get the error:

```
Software Version: 12.900.10.16
PPD file '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/12.5/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.ppd' has been loaded successfully.
DESIGN : FlashPro6;  CHECKSUM : A8EA;  PDB_VERSION : 1.0
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
programmer 'S20053Q3C8' : FlashPro5
Opened '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/12.5/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.pro'
￼Warning: FlashPro5 TCK was set to 4000000, but has been reduced to 30 to meet the minimum max TCK of the chain.
programmer 'S20053Q3C8' : Scan Chain...
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
ERROR!!! Failed to read voltage supplies.
￼Error: programmer 'S20053Q3C8' : FP5 SetOperationStatus: SPI_SetHiSpeedDeviceGeneralPurposeInputOutputPins
         FP5: Error code = 0 -
￼Error: programmer 'S20053Q3C8' : Scan Chain FAILED.
￼Error: Failed to run Action.
Exported log file /home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/12.5/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.log.
```

Nonetheless, I'm able to write the "SoC-FPGA + zero-stage bootloader FlashPro Express programming file" into the polarfire without issues using the embedded flashPro 6 hardware and the FlashPro Express 12.5 software.

Therefore, I'm unable to use the embedded flashPro 6 with modern FlashPro Express software because (it seems) that I cannot update the eFP6_revB_TAG_v4_ppd.job. Is this file compatible with my icicle kit? Is there something that I'm missing?

Thank you!

hughbreslin, 2020-12-22:
Hey @aleixrocks have another look at the image in the [procedure doc](https://download-soc.microsemi.com/FPGA/v12.5/sp1/README.pdf) - you need to use the header beside the one you're using to program the FPGA to program the eFP6 🙂 

The error you're getting is saying its found the wrong part, the eFP6 uses a SmartFusion2 which is the M2S025 device its unable to find.

aleixrocks, 2020-12-22:
Hi @hughbreslin! Thank you for the quick response and the explanation of the error.

Indeed I was using the wrong header :man_facepalming: . However, although no more errors are shown in the FlashPro Express software, I'm still unable to write the `eFP6_revB_TAG_v4_ppd.job` job. FlashPro Express gets stuck with 0% progress for more than 20 minutes as shown below. The external FLashPro5 hardware "Activity" led blinks.

```
Software Version: 12.900.10.16
PPD file '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/12.5/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.ppd' has been loaded successfully.
DESIGN : FlashPro6;  CHECKSUM : A8EA;  PDB_VERSION : 1.0
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
programmer 'S20053Q3C8' : FlashPro5
Opened '/home/aleix/bsc/projects/riscv/boards/polarfire/flashproexpress_projects/12.5/update_efp6_revB/eFP6_revB_TAG_v4_ppd/eFP6_revB_TAG_v4_ppd.pro'
￼Warning: FlashPro5 TCK was set to 4000000, but has been reduced to 30 to meet the minimum max TCK of the chain.
programmer 'S20053Q3C8' : Scan Chain...
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 1 MHz
programmer 'S20053Q3C8' : Check Chain...
programmer 'S20053Q3C8' : Scan and Check Chain PASSED.
programmer 'S20053Q3C8' : device 'M2S025' : Executing action PROGRAM
Programmer: 'S20053Q3C8' : JTAG TCK / SPI SCK frequency = 0 MHz
programmer 'S20053Q3C8' : device 'M2S025' : Family: SmartFusion2
programmer 'S20053Q3C8' : device 'M2S025' : Product:      M2S025
programmer 'S20053Q3C8' : device 'M2S025' : Cortex-M3 Enabled (M3_ALLOWED = 1)
programmer 'S20053Q3C8' : device 'M2S025' : EXPORT ISC_ENABLE_RESULT[32] = 007c6a44
programmer 'S20053Q3C8' : device 'M2S025' : EXPORT CRCERR[1] = 0
programmer 'S20053Q3C8' : device 'M2S025' : EXPORT EDCERR[1] = 0
programmer 'S20053Q3C8' : device 'M2S025' : Programming FPGA Array and eNVM...
``` 

I have tried using the same jumper configuration as in the procedure doc. I have tried with both FlashPro Express 12.5 and 12.6. Might you have some hint on what I'm missing?

nitindeshpande, 2020-12-22:
Can you please check if jumper J9 and J24 are closed?

aleixrocks, 2020-12-22:
Yes, both J9 and j24 are closed. I'm attaching a picture of my setting. Also, I made sure that the small arrow on the blue connector of the ribbon cable plugs on the arrow printed on the pcb of the header.

[image]

aleixrocks, 2020-12-23:
After the failed attempts to upload the `eFP6_revB_TAG_v4_ppd.job`, the embedded FlashPro 6 is no longer being recognized by the kernel and, therefore, not usable by the FlashPro Express software.

```
$ dmesg
[...]
[ 5470.693463] usb 2-4: new full-speed USB device number 14 using xhci_hcd
[ 5470.853384] usb 2-4: device descriptor read/64, error -71
[ 5471.123406] usb 2-4: device descriptor read/64, error -71
[ 5471.403391] usb 2-4: new full-speed USB device number 15 using xhci_hcd
[ 5471.563401] usb 2-4: device descriptor read/64, error -71
[ 5471.843398] usb 2-4: device descriptor read/64, error -71
[ 5471.963467] usb usb2-port4: attempt power cycle
[ 5472.673382] usb 2-4: new full-speed USB device number 16 using xhci_hcd
[ 5472.673543] usb 2-4: Device not responding to setup address.
[ 5472.903544] usb 2-4: Device not responding to setup address.
[ 5473.133383] usb 2-4: device not accepting address 16, error -71
[ 5473.283358] usb 2-4: new full-speed USB device number 17 using xhci_hcd
[ 5473.283506] usb 2-4: Device not responding to setup address.
[ 5473.503507] usb 2-4: Device not responding to setup address.
[ 5473.723351] usb 2-4: device not accepting address 17, error -71
[ 5473.723447] usb usb2-port4: unable to enumerate USB device
```

Also, I'm no longer sure if this question is relevant to this repository. Maybe I should move it somewhere else?

nitindeshpande, 2020-12-23:
I think that's right, this issue is not related to reference design.
Please contact the customer support and we will address this separately.

aleixrocks, 2020-12-23:
I will, thanks!

closed 2020-12-23 by aleixrocks