polarfire-soc/icicle-kit-reference-design #4: Synthesis error - Reference to undefined module CAXI4DMAOI11I

rroulleau, 2021-02-16:
Using Libero v12.6
When I Place&Route the project in Libero I have the following error in synthesis:
`@E: CG389 :"C:\<localdir>\microsemi\Polarfire\icicle-kit\icicle-kit-reference-design-master\MPFS_ICICLE_eMMC\component\Actel\DirectCore\COREAXI4DMACONTROLLER\2.0.100\rtl\vlog\core_obfuscated\coreaxi4dmacontroller_dma_start_ctrl.v":217:0:217:12|Reference to undefined module CAXI4DMAOI11I`
 Do I need to install as well the MegaVault 12.6 local on my machine?

hughbreslin, 2021-02-16:
Hey @rroulleau hmm no you shouldn't need to - I suspect the issue here is path length as you're running on Windows. The path above (including <localdir>) is 217 characters long, I'm not sure what path <localdir> is referencing but it could be pushing you above the 260 limit and cause the files not to be found. 

Can you try re-generating the design (moving your existing project won't create these files 🙁) in a directory with a shorter path? I normally save these projects in something like "D:\PF_SOC\test_prj_x" to keep the initial path short.

rroulleau, 2021-02-16:
Ok, moved the project to a shorter path location and it completed the place&route,
Thanks,

closed 2021-02-16 by rroulleau