polarfire-soc/icicle-kit-reference-design #5: How to access registers mapped in the fabric with a linux command line ?

rroulleau, 2021-02-16:
Can I set the following register SDIO_register, located at address 0x4f00_0000, via the linux prompt or do I need to use softconsole?
As a more generic question, can I access registered mapped in the fabric using a command like:
# devmem  0x4f00_0000 0x1 
Thanks,

hughbreslin, 2021-02-16:
Linux can access this register, your example devmem command looks correct if using Buildroot, Yocto includes devmem2 so something like:

devmem2 0x4f000000 w 0x1

should work 🙂 Just keep in mind that setting / unsetting this register will swap your access between eMMC and SD so you could lock up your system by changing it on the go.

rroulleau, 2021-02-16:
Ok, I do not have the linux devmem2 command on my current dev board. I guess I need update my linux, which I started to do  but using the SD card. This is why I was trying to flip this bit in the fabric... I will update first the Linux in the eMMC.

hughbreslin, 2021-02-16:
Hmm ok, I'm a bit confused by "I started to do but using the SD card. This is why I was trying to flip this bit in the fabric..." as the latest HSS versions can set the bit on power on and determine if an SD card is inserted and if its not it'll clear the bit and mount eMMC so you shouldn't really need to mess with it. If you want the latest HSS without building it the programming file in the releases of this repo also programs the eNVM with the HSS

rroulleau, 2021-02-16:
Ha ok, as the board boot either way (SD card connected or not) I was not sure which linux image was booting. I've already started to update the eMMC. I will check after.
The "devmem2" and other script mentionned in this repo such as /opt/microchip/fpga-fabric-interfaces/lsram are in the rootfs correct? so if I update the eMMC using the *.wic image I should be up-to-date with this repo?

Thanks for the explanation, I was not grasping the role of HSS at boot time, just started reading the Microchip_PolarFire_SoC_Software_Development_and_Tool_Flow_User_Guide_V1.pdf

hughbreslin, 2021-02-16:
Hmm I'll need to double check - you may need to build the full development tools image to include devmem2 (the releases *.wic files are the core-image-minimal build). If you're building the full image devmem2 is definitely there if you're using the provided image I'll follow up. 

Ahh that doc may need an update with this info as it could have been created before the switching feature was added. You might be interested in this [training series](https://www.microchip.com/training/webinars/fpgas/risc-v-innovation-unleashed-training-series/risc-v-innovation-unleashed-training-archive) around PolarFire SoC if you haven't seen it :)

hughbreslin, 2021-02-17:
Hey @rroulleau devmem2 etc aren't in the minimal dev images you'll need to build the the full development image to get them but yes once you update with the .wic image you should be good to go :)

rroulleau, 2021-03-03:
My HSS version on my icicle kit is the following:
-------------------------------------------------------
- HSS Version
-------------------------------------------------------

[23.227496] HSS_TinyCLI_Parser(): >> VERSION
[27.798436] HSS_E51_Banner(): PolarFire(R) SoC Hart Software Services (HSS) - **version 0.99.10**
(c) Copyright 2017-2020 Microchip Corporation.

[27.813105] HSS_E51_Banner(): incorporating OpenSBI - version 0.6
(c) Copyright 2019-2020 Western Digital Corporation.

[27.825865] HSS_PrintBuildId(): Build ID: d6953091fa5556fb5585e6efbd29371ec2fa5c59
[27.835066] HSS_PrintToolVersions(): Built with the following tools:
 - riscv64-unknown-elf-gcc (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 8.3.0
 - GNU ld (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 2.32

Does it support the SD boot mode? from the HSS menu it does not look likely and even with SD card plugged in it only boot from eMMC, can you confirm? I will look at updating the HSS on my board.

hughbreslin, 2021-03-03:
Ahhh ok that probably won't help you much - if you want to save building the HSS and going through that flow you can use the programming file [here](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2021.02) - we're providing the HSS as an eNVM client with the design releases so programming with this file will get you up to the latest version. 🙂

rroulleau, 2021-03-03:
I got the SD card booting linux. HSS update did help indeed. 
Using devmem I can access the axi bus connected to the fabric (lsram or sdio_reg) but can devmem access the APB bus from FIC_3? I am trying to flip the LEDs by writing at address 0x42000000 but it does not seem to work.

hughbreslin, 2021-03-03:
Excellent :) Yes devemem can access FIC 3 (sdio_register is on this FIC so you've already done it), The LEDs are connected to MSS GPIO 2 though and won't be accessible at 0x42000000 - you could use sysfs you should be able to set / unset the LEDs and the GPIO driver or do it by hand using devmem, check out this [video](https://www.youtube.com/watch?v=88xk1dV3jDs&list=PL9B4edd-p2agS6QkXUZjAvjlPiyAO5RP0&index=6) on how this was done using the Yocto image to set and clear the GPIOs

rroulleau, 2021-03-04:
Thanks for the video link. The HSS update seems to have solve my issues.

closed 2021-03-04 by rroulleau