polarfire-soc/icicle-kit-reference-design #6: icicle kit - AXI & APB clock alternative?

rroulleau, 2021-03-04:
With the icicle kit eval board, I am looking at updating the RTL fabric design by removing the PCIe interface PCIE_BASE IP.  As this PCI_BASE IP is generating the AXI_CLK_125MHZ and the APB_CLK_62_5MHZ, what would be your recommandation to replace those 2 clocks on this icicle board?

hughbreslin, 2021-03-04:
@rroulleau I've got a few updates queued up internally that I can hopefully get out in a few days - one of them is a clock and reset clean up which moves all clock and reset generation into its own smart design separately to the PCIe block. If you can give me a few days to push this out it'll make life easier for you doing this. 

If you can't wait you could remove the PCIe components while leaving the clock generation as is in the PCIe block, it won't be as clean but it'll work :)

rroulleau, 2021-03-04:
Cool, let me know when it is ready!

hughbreslin, 2021-03-05:
@rroulleau The latest updates are pushed 🥳 If you generate the design from master now you'll be able to see the separated PCIe and clocks :)

To remove it from the system fully you'll have to do something along these lines (I haven't tried it yet so I may have missed a step):
1. Update the design 
2. build a custom HSS removing the PCIe init service (this will crash the HSS otherwise)
3. Remove the dts entry for PCIe from Yocto / Buildroot

Hopefully thats the bulk of it, let me know if you run into a problem

rroulleau, 2021-03-18:
@hughbreslin, I used your updated design, removed the PCI module but in the new CLOCKS_AND_RESETS module  the TRANSMIT_PLL_0 requires to connect the CLKS_TO_XCVR bus to "something" otherwise it complains in the compilation (mark it as unused do not work). In your design it is connected to the PCI module (CLKS_FROM_TXPLL_TO_PCIE) but as I removed  it, I have nowhere to connect it. How can I worked around this problem to ensure a 125 MHz clock?

hughbreslin, 2021-03-18:
Ahh I hadn't realized that would happen :( What you could do is instantiate a CCC block and connect that to the 160MHz Oscillator output and use it to generate the 125MHz clock and the 62.5MHz clock, if you did this you could remove the PCIE_REF_CLK, GLITCHLESS_MUX, clock dividers, PCIE AND2 instance and TRANSMIT_PLL and have a clean set up. I've attached a screenshot - my CCC is configured with a 160MHz input clk, I de-selected "Expose PowerDown Port" and output clock 0 is 125MHz and output clock 1 is 62.5MHz. I've left the "PCIe_x" and "CLKS_TO_XCVR" disconnected in the screenshot as this is what I think you need to get rid of - in you case obviously just delete them :) 
[image]

rroulleau, 2021-03-18:
Ok thanks it looks good now :-). Just a question on the OSCILLATOR_160MHz should I select "RCOSC_160MHz to Global" or leave it to "RCOSC_160MHZ to Clock Divider"?

hughbreslin, 2021-03-18:
Nice spot! Global is the default and makes the most sense here 👍

rroulleau, 2021-03-19:
@hughbreslin coming back to the steps to follow, on 2) rebuild a custom HSS wihtout the PCIe init service -  in the make menuconfig I only see a reference in (Top)->general configuration options -> Memory options -> Setup PCIe. Is it there another one as I do not see any item under "Services"
Thanks.

hughbreslin, 2021-03-19:
The one you have it correct, I knew it existed but thought it was in as a service.

` (Top)->general configuration options -> Memory options -> Setup PCIe` - this is the correct option to disable as it will try to use the DRI you removed to configure some registers, if the peripheral doesn't exist the system will hang :)

hughbreslin, 2021-07-28:
Hey @rroulleau are you ok to close this issue?

rroulleau, 2021-07-29:
sure, thanks again.

closed 2021-07-29 by rroulleau