polarfire-soc/icicle-kit-reference-design #9: AXI4 DMA controller DirectCore version 2.1.102 undefined modules

mpsitech, 2021-07-20:
Synthesis fails on a brand new installation of Libero SoC v2021.1 on Windows 10:

@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_ram_1k20_wrapper.v":111:2:111:18|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_bd
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_int_ext_dscrptr_cache.v":188:2:188:17|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_dscConCacheNM
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_int_ext_dscrptr_cache.v":206:2:206:17|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_dscCacheNM
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_int_ext_dscrptr_cache.v":224:2:224:17|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_dscCacheM
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\CoreAXI4DMAController_int_0_ControllerFIFO.v":105:2:105:22|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_fifo_0
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_cache.v":188:2:188:22|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_cache
@E: CG389 :"C:\Users\mpsitech\fpgacode\wskd_tmp1\MPFS_ICICLE_SD_CARD\component\work\DMA_CONTROLLER\DMA_CONTROLLER_0\rtl\vlog\core\coreaxi4dmacontroller_cache.v":206:2:206:22|Reference to undefined module DMA_CONTROLLER_DMA_CONTROLLER_0_ram_cache

The latest tagged release, 2021.04, which uses version 2.0.100 of the DMA DirectCore works without problems, so does manually reverting ICICLE_KIT_SD_CARD.tcl and DMA_CONTROLLER.tcl.

MarkusPLM, 2021-07-21:
> Synthesis fails on a brand new installation of Libero SoC v2021.1 on Windows 10:
>
> @e: CG389 [...]

The same happened to me. Further to this, when I try to save the project, Libero reports: "The die used in this project is not available with your configured license and tools version." After clicking OK, an error message appears saying: "Unable to select die 'MPFS250T_ES'. The die value must be one of the following: 'MPFS250T_ES MPFS250TS_ES ....'. I'm using a silver license.

mpsitech, 2021-07-21:
That latter part, at least for me, could be fixed by closing and re-opening Libero SoC.

MarkusPLM, 2021-07-21:
> 
> 
> That latter part, at least for me, could be fixed by closing and re-opening Libero SoC.

Yes, this worked for me as well. I'm just not sure, whether the project is saved correctly then as Libero reported something like "could not properly save project"

hughbreslin, 2021-07-21:
Hi @mpsitech and @MarkusPLM thanks for reporting this issue - it appears to only be affecting silver licenses with Libero. I don't have a resolution at the moment (apart from the workaround from @mpsitech). Will investigate further and get back to you.

MarkusPLM, 2021-07-26:
Hi @hughbreslin,
are there any news about this? 
I was able to create a simple example from scratch using Libero only, so this works with silver license. But not being able to synthesize the reference design stops me from evaluating the ICICLE properly...

hughbreslin, 2021-07-26:
Hey @MarkusPLM we'll be reverting the DMA core to v2.0.100 for the next release which is due shortly, in the mean time you could use the [2021.04 release](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2021.04) which has the older DMA core in it and will synthesize the design with a silver license.

MarkusPLM, 2021-07-26:
> Hey @MarkusPLM we'll be reverting the DMA core to v2.0.100 for the next release which is due shortly, in the mean time you could use the [2021.04 [...]

Thank yoy!

hughbreslin, 2021-08-09:
Hi @MarkusPLM @mpsitech this issue should be resolved in the [latest release](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2021.08)

closed 2021-08-16 by hughbreslin