polarfire-soc/icicle-kit-reference-design #17: icicle-kit-reference-design-2022.08.zip build failed on Libero V2022.2

zizimumu, 2022-08-26:
i downloaded latest released icicle-kit-reference-design-2022.08.zip, used Libero V2022.2.0.10 to execute script to generate the project, but when execute "generate bitstream", Libero shew following error, "C:\Microchip\Libero_SoC_v2022.2\SynplifyPro\bin\synplify_pro.exe" is not found, i check out the files in C:\Microchip\Libero_SoC_v2022.2\SynplifyPro\bin, actually there is no "synplify_pro.exe"

+++++++++++++++++++++++++++++++++++++++++++
The MPFS_ICICLE project was opened.
￼INFO: The option "Abort flow if errors are found in SDC" is turned ON in Project Settings> Design flow page. The Synthesis tool will fail if errors are found in associated SDC files. Please uncheck the option to ignore the errors and continue running the tool

￼Info: Active implementation is 'synthesis'
Starting Synplify Pro ME...
￼Error: Failed to execute: 'C:\Microchip\Libero_SoC_v2022.2\SynplifyPro\bin\synplify_pro.exe'.
         Executable cannot be found, you may need to reinstall this tool.
￼Error: Unable to start Synplify Pro ME.
         From the Project menu, select 'Tool Profiles...' and change the Synplify Pro ME profile.
￼Error: Synthesis failed.

ryandormody, 2022-08-26:
I had the same problem, the default Libero tool profile for synthesis uses an incorrect path.  I created a new Synthesis Tool Profile (In the Libero GUI, under the project / tool profile pulldown) and pointed to C:\Microchip\Libero_SoC_v2022.2\SynplifyPro\bin64\mbin\synplify.exe.

zizimumu, 2022-08-27:
yes, add a new Tool Profile point to C:\Microchip\Libero_SoC_v2022.2\SynplifyPro\bin64\mbin\synplify.exe can work.

closed 2022-09-02 by zizimumu