polarfire-soc/icicle-kit-reference-design #18: issue executing the script in Ubuntu

controlpaths, 2022-10-08:
hello, 

I am trying to generate the reference design with the release 2022.9 and I have the following error.

`Info: Core 'Actel:SgCore:PF_OSC:1.0.102' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_CCC:2.2.214' was successfully downloaded.
Info: Core 'Actel:DirectCore:CORERESET_PF:2.3.100' was successfully downloaded.
Info: Core 'Microsemi:SgCore:PFSOC_INIT_MONITOR:1.0.304' was successfully downloaded.
Info: Core 'Actel:DirectCore:COREAXI4INTERCONNECT:2.8.103' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_CLK_DIV:1.0.103' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_DRI:1.1.104' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_NGMUX:1.0.101' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_PCIE:2.0.116' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_TX_PLL:2.0.300' was successfully downloaded.
Info: Core 'Actel:SgCore:PF_XCVR_REF_CLK:1.0.103' was successfully downloaded.
Info: Core 'Actel:DirectCore:CoreAPB3:4.2.100' was successfully downloaded.
Info: Core 'Actel:DirectCore:COREAXI4DMACONTROLLER:2.0.100' was successfully downloaded.
Info: Core 'Actel:DirectCore:CoreGPIO:3.2.102' was successfully downloaded.
Info: Core 'Actel:SystemBuilder:PF_SRAM_AHBL_AXI:1.2.108' was successfully downloaded.
Info: Core 'Actel:Simulation:CLK_GEN:1.0.1' was successfully downloaded.
Info: Core 'Actel:Simulation:RESET_GEN:1.0.1' was successfully downloaded.
Info: Core 'Actel:DirectCore:corepwm:4.5.100' was successfully downloaded.
Info: Core 'Actel:DirectCore:COREI2C:7.2.101' was successfully downloaded.
Info: Core 'Actel:DirectCore:CoreUARTapb:5.7.100' was successfully downloaded.
Info: Design 'ICICLE_MSS' was successfully imported.
Info: Component 'CLK_DIV' was successfully created and configured with vendor:Actel library:SgCore name:PF_CLK_DIV version:1.0.103.
Info: Component 'CORERESET' was successfully created and configured with vendor:Actel library:DirectCore name:CORERESET_PF version:2.3.100.
Error: Generator 'ActSpiritHdl' for instance 'GLITCHLESS_MUX_0' of 'Actel:SgCore:PF_NGMUX:1.0.101' failed.
Error: Unable to generate design "GLITCHLESS_MUX" due to errors.
Generation of hardware IP core(s) failed.

[OK]
Error: Could not configure component 'GLITCHLESS_MUX'.
Error: The command 'create_and_configure_core' failed.
Error: Failure when executing Tcl script. [ Line 160 ]
Error: The Execute Script command failed.
`
I am using Libero 2022.2 in Ubuntu 20.04.5, and the script is executed without arguments. Also, on the first try, I have executed the script with the argument MSS_BAREMETAL with the same result.

Thanks

hughbreslin, 2022-10-11:
Hey @controlpaths - this issue is more a usage question for a discussion.

Have you run the additional packages check script for Libero to ensure you have all of the required packages installed? I sanity checked the script this morning on my Linux machine (Ubuntu 20.04.5 LTS) and it generates the design ok so I suspect its a missing package for Libero or something similar.

To test you could create an empty design and try to instantiate an "PF_NGMUX" component to see if you can generate it standalone.

hughbreslin, 2022-10-18:
Hey @controlpaths did you make any progress on this? :)

hughbreslin, 2022-11-03:
Closing this issue as there hasn't been an update for a few weeks - please feel free to reopen it if you have any update on this and are still seeing the issue

closed 2022-11-03 by hughbreslin