polarfire-soc/icicle-kit-reference-design #20: SoC Icicle Kit Reference Design Generation Tcl Script doesn't work with Libero® SoC v2022.3

bekulio, 2022-12-25:
SoC Icicle Kit Reference Design Generation Tcl Script doesn't work with Libero® SoC v2022.3

The automatic Design Generation will start OK but then after some time Libero will crash without leaving any clue or log to the problem it encountered. It was tested several times in windows 11, always ending in a sudden crash, before the generation got finished.

Once I installed Libero® SoC v2022.2 it worked as expected, and the design was generated.

hughbreslin, 2023-01-06:
Hi @bekulio apologies for the delay in getting back, windows 11 isn’t a supported operating system for libero, the available platforms are available [here](https://www.microchip.com/en-us/products/fpgas-and-plds/fpga-and-soc-design-tools/fpga/libero-software-later-versions).

hughbreslin, 2023-01-09:
Hi @bekulio I tested this on a Windows 10 machine and the design generates without an issue using Libero v2022.3. Closing this as it is a platform support issue rather than an issue with the design itself. Please feel free to open a [tech support](https://microchipsupport.force.com/s/) case for Windows 11 support in Libero.

closed 2023-01-09 by hughbreslin