polarfire-soc/icicle-kit-reference-design #21: Error: Failure when executing Tcl script. [ Line 96: ERROR: Invalid parameter entry "MSS_MANUAL_DDR_PHY_OFFSET_ENABLE true"

Kian-KH, 2023-03-15:
Hi,
When executing the provided Tcl script "MPFS_ICICLE_KIT_REFERENCE_DESIGN.tcl", a failure error occurs regarding MSS_MANUAL_DDR_PHY_OFFSET_ENABLE defined in script_support/MPFS_ICICLE_MSS_linux.cfg, line 1411:
https://github.com/polarfire-soc/icicle-kit-reference-design/blob/master/script_support/MPFS_ICICLE_MSS_linux.cfg#L1411
Which could be solved by commenting/removing that line.
[image]

hughbreslin, 2023-03-15:
Hi @Kian-KH what version of libero are you using to run the tcl script? You’ll need 2022.3 to run the latest version

Kian-KH, 2023-03-15:
Ah, I'm using 2022.2. Thanks for your response, @hughbreslin !

closed 2023-03-15 by Kian-KH