polarfire-soc/icicle-kit-reference-design #22: Accessing the fabric memories

cchalou98, 2023-03-27:
[image]
Hello everyone, 
I keep having this issue trying to upload hex file into the LSRAM of the fabric, could anyone give me steps to follow cuz I think am missing something.

hughbreslin, 2023-03-27:
Hey @cchalou98 is this a hex file generated with SoftConsole? You may need to normalize the addresses in the hex file, its described in [this section](https://mi-v-ecosystem.github.io/SoftConsole-Documentation/SoftConsole-v2022.2-RISC-V/using_softconsole/projects.html?highlight=lma#deploying-release-hex-files-for-production) of the SoftConsole release notes :)

cchalou98, 2023-03-27:
> Hey @cchalou98 is this a hex file generated with SoftConsole? You may need to normalize the addresses in the hex file, its described in [this [...]

No, actually not from SoftConsole

hughbreslin, 2023-03-27:
Hmm ok, my suggestion would still apply if you have an [extended segment address record](https://developer.arm.com/documentation/ka003292/latest) in your hex file, Libero can't process these as the data shouldn't be offset in the memory. Without knowing how your generating your hex file or maybe how big it is it hard to help more, we should probably make this a discussion item.

cchalou98, 2023-03-27:
> Hmm ok, my suggestion would still apply if you have an [extended segment address record](https://developer.arm.com/documentation/ka003292/latest) in your hex file, Libero can't process these as the [...]

I am just importing a hex file into the memories in Libero, and am not sure that's the way to do it, could you please provide me with good documents on how to do that?

hughbreslin, 2023-04-14:
Hi @cchalou98 sorry about the delay in getting back - are you still facing this issue? I'm not sure what you want documentation on, is it importing clients to memory or how to generate a client?

closed 2026-05-14 by vfalanis