polarfire-soc/icicle-kit-reference-design #24: Offline execution of ICICLE reference kit TCL script

karthyyy, 2024-04-24:
Hi,

  I am trying to get familiarized with ICICLE Kit with the reference design. I am working in a PC which is offline (not connected to internet). So I downloaded MegaVault and assigned the location. When I try to run the script in Libero, its throwing following error. The code is trying to download some cores but since there is no internet connection, it cannot download and returning errors. However these cores are visible from the Libero catalog. Is there any other methods to run the script successfully ?

TCL_BEGIN: D:\...\icicle-kit-reference-design-master\MPFS_ICICLE_KIT_REFERENCE_DESIGN.tcl
Project path length ok.
no command line argument passed
Creating a new project
Info: This version of Libero supports only the enhanced constraint flow.
The MPFS_ICICLE project was created.
Downloading Actel:SgCore:PF_OSC:...
INFO:Could not download the core Actel:SgCore:PF_OSC:
Error: The command 'download_core' failed.
Downloading cores failed, the script will continute but will fail if all of the required cores aren't present in the vault.
Error: Failure when executing Tcl script. [ Line 96: child killed: unknown signal ]
Error: The Execute Script command failed.

Regards
   Karthik R

bekulio, 2024-04-24:
Hi, I had this issue in the beginning as well. Currently, I’m in vacation only with my phone so I cannot give examples. Basically, you need to edit the main tcl and comment out IP downloading.

karthyyy, 2024-04-25:
Hi @bekulio ,

Thanks for the response. I had tried removing the "download" command in the script but its still throwing the error,

```
Error: Failure when executing Tcl script. [ Line 96: child killed: unknown signal ]
Error: The Execute Script command failed.
```
   It will be of much help if you can point out the edit in the tcl script. I shall wait. Till then I will try with some other examples.

Regards
  Karthik R

bekulio, 2024-04-25:
Hi, did you remove all the lines with download  core? Because it looks like you removed only part of the line.