polarfire-soc/icicle-kit-reference-design #28: Latest Icicle Release and using PCIE

haroldnelson-2021, 2024-11-22:
I am debugging a PCIE issue using the Icicle board.  

- For one setup I connected a StarTech PEX1394B3 mini-PCIE card to the Icicle, and built a Linux image with the drivers for it.
- For another setup (that previously worked with the Icicle a couple of years ago), I connected a TP-Link TG-3468 mini-PCIE card.

For the TG-3468, the "lspci" command seems to return mostly good information except all of the memory is "disabled".  Can you offer any insights for the below?  Thanks in advance.

root@icicle-kit-es:~# lspci -v
00:00.0 Unclassified device [0006]: Actel Device 1556 (prog-if 04)
        !!! Invalid class 0006 for header type 01
        Flags: fast devsel
        Memory at <unassigned> (64-bit, prefetchable) [disabled]
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 08001000-08001fff [size=4K]
        Memory behind bridge: 18000000-181fffff [size=2M]
        Prefetchable memory behind bridge: 0000000009000000-00000000091fffff [size=2M]
        Capabilities: [80] Express Root Port (Slot+), MSI 00
        Capabilities: [e0] MSI: Enable- Count=8/8 Maskable- 64bit+
        Capabilities: [f8] Power Management version 3
        Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [200] Advanced Error Reporting

01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (TP-Link TG-3468 v4.0 Gigabit PCI Express Network Adapter)
        Flags: fast devsel, IRQ 255
        I/O ports at 1000 [disabled] [size=256]
        Memory at 3018004000 (64-bit, non-prefetchable) [disabled] [size=4K]
        Memory at 3018000000 (64-bit, non-prefetchable) [disabled] [size=16K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 01
        Capabilities: [b0] MSI-X: Enable- Count=4 Masked-
        Capabilities: [d0] Vital Product Data
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [140] Virtual Channel
        Capabilities: [160] Device Serial Number 01-00-00-00-68-4c-e0-00
        Capabilities: [170] Latency Tolerance Reporting
        Capabilities: [178] L1 PM Substates

jamiegibbons, 2024-11-27:
Hi @haroldnelson-2021,

Could you tell us which reference design version and which software & software versions you are using please? Including what build system and release/version you are using. Could you also tell us any custom configs you have?

Without this information is it very hard for us to give much insight. With this information, we can give you a better informed answer.

Looking forward to your response.

Thanks,
Jamie.

haroldnelson-2021, 2024-11-27:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi Jamie,

I’m adding James Schaffer as he did the non-GitHub Linux builds for the FireWire NIC and for his WiFi NIC.

For the TP-Link PCIE-Ethernet NIC I used the default GitHub FlashPro and Linux files from the latest release which I think is 2024.09.

It seems all configurations are failing the same way, i.e. with “memory disabled”.  Also, I noticed a DMESG status saying “DLUP exit” at the 4-5 second timeline after PCIE enumeration.

Regards,
Harold

Harold Nelson
(919) 349-9356
Embedded Solutions Engineer, FPGA/SOC Products
***@***.******@***.***>

haroldnelson-2021, 2024-12-02:
James, can you describe how you generated the Linux images we used for our three PCIE NIC boards that seem to not enumerate the PCIE interfaces, completely?

Thanks.

Regards,
Harold

Harold Nelson
(919) 349-9356
Embedded Solutions Engineer, FPGA/SOC Products
***@***.******@***.***>

haroldnelson-2021, 2024-12-02:
The Linux images were built using v2024.09 Yocto BSP with a couple of changes to the kernel configuration to enable drivers for the cards.  I’ll attach my kernel .config file to this email for reference.

I did not make any changes to the default Yocto recipes.

Best regards,
--James

James Schaffer
Principal Embedded Solutions Engineer
---------------------------------------------------------------------
Microchip Technology Inc.
7901 Strickland Rd. Suite 101, Raleigh, NC 27615
Office: (+1) (919) 846-3383
***@***.******@***.***>  www.microchip.com<http://www.microchip.com/>
[HD:Users:jeanette:Desktop:MCHP Email Logo.jpg]  ***@***.*** <https://secure.microchip.com/mu>

Importance: High

jamiegibbons, 2024-12-03:
Hi Harold, hi James, 

Thank you for the information you have provided. However, we are still unclear on which information is for which NIC. Could you clarify which failures/logs and which config files are for which of the two NICs? If possible, can you provide the logs and kernel configs from both NICs?

We suggest that the best way to provide us with the actual kernel config file used during testing is to provide us with the output of the following command: `zcat /proc/config.gz`

We feel it is also worth us mentioning that we have confirmed a StarTech NIC and a TP-Link NIC are both working with our latest 2024.09 release. Our NIC are not the same model as either of yours, however, it might be worth noting the following. 
For our wired StarTech model NIC, the required kernel kconfig option is `CONFIG_R8169=y`. The following commands can be used to bring it up and test: `ifconfig eth2 down`, `ifconfig eth2 up`, `udhcpc -i eth2`, `ping www.google.com`. You can use `ifconfig` to find the name associated with your device.
For our wireless TP-Link model NIC, the required kernel kconfig options are `CONFIG_CFG80211=y`, `CONFIG_MAC80211=y`, `CONFIG_RFKILL=y`, `CONFIG_RTL8188EE=y`. You can then use WPA Supplicant and DHCP Client to bring up the connection with the following commands: `wpa_supplicant -B -i wlp1s0 -c /etc/wpa_supplicant.conf`, `ifconfig wlp1s0 up`, `dhclient -r`, `dhclient wlp1s0`, `ping -c 4 www.google.com`. Again, using `ifconfig` or `ip a` to find the name of your device. Please note: you must edit your "wpa_supplicant.conf" file with your wireless network details.

Please let us know if this is helpful and provide us with the details mentioned above.

Thank you,
Jamie